esa/asn1scc #281: [SCALA] wrong method call

final87, 2023-11-22:
Creating Scala code from an asn1 definition with fixed size array, like this:
```scala
IntermediateDownlinkPartReport ::= SEQUENCE {  
     packetPart                                  SEQUENCE (SIZE (255)) OF INTEGER(0..255)  
}
``` 

in the Def file the class of the array is generated with a constructor accepting one parameter:
```scala
case class IntermediateUplinkPart_packetPart (arr: Array[IntermediateUplinkPart_packetPart_elem])  
{
    require(arr.length == 255)
}
``` 

but then the object is created passing two parameters
```scala
var pVal: IntermediateUplinkPart_packetPart = IntermediateUplinkPart_packetPart(0, Array.fill(255){0})`
``` 

which causes compilation error.

deiruch, 2023-11-28:
Thank you for the bug report. We'll take a look at it.

fschramka, 2023-12-01:
Hi @final87 

Thanks for letting us know :) This simple case was not covered by the PUS-C asn1 input.

We fixed it (and other stuff) in: 
https://github.com/ateleris/asn1scc/tree/scala-backend
 
It is only a draft right now :) Please use this branch until our work is pulled into the main repo :)

closed 2024-02-19 by usr3-1415