esa/asn1scc #286: C global constant variable name collision

dsabala, 2023-12-11:
Hi

I am using asn1scc with following arguments:
```
asn1-docker.sh \
--acn-enc \
--auto-test-cases \
--icduper pus/docs/autogenerated/index.html \
--field-prefix AUTO \
--type-prefix pp_ \
--c-lang \
-o $directory_output \
$directory/*.asn1 \
$directory/*.acn
```

Types as well as function names are prefixed with `pp_` which solves names conflicts I have due to similarity between two devices asn1 codes. However I still get one name collision between global variables:

Related *.asn1
```asn1
some_constant INTEGER ::= 1
```

Generated header file
```c
extern const asn1SccSint some_constant;
```

Two issues:
- consider including global, constant variables to be prefixed with `--type-prefix`
- consider extending `--type-prefix` help message to include information, that function names are prefixed too

Best regards

usr3-1415, 2024-01-02:
Hi,

Thank you for your detailed feedback and suggestions regarding the asn1scc arguments and the naming conflicts you've encountered.

I agree that extending the --type-prefix argument to include global, constant variables seems like a viable solution. However, this change could potentially break existing code that relies on the current naming conventions.

@maxime-esa, could you please share your thoughts on this matter? If we proceed with this update, we'll need to consider its impact on existing users. Do you think we should implement this change?

maxime-esa, 2024-01-03:
@usr3-1415 that would indeed break existing code (in C), so I do not think it is a good thing to add this option without any mean to disable it.

Alternatively, a new `--prefix-with-module-name` option that would be mutually exclusive with `--type-prefix` could be introduced (or `--type-prefix MODULE`).

All types and constants would be prefixed the module name (`Module-Name DEFINITIONS ::= ...`) avoiding all risks of collisions. In essence the idea (as with type prefixes) is to create a level of namespace in C (not needed in Ada as types are grouped in packages named with the namespace already).

However this would probably complicate the String templates quite a bit.