esa/asn1scc #288: ACN optional choice with determinant

mario-bucev, 2024-01-16:
For the following definitions:
```
Determinant3 DEFINITIONS AUTOMATIC TAGS ::= BEGIN
EXPORTS ALL;

MyChoiceEnum ::= ENUMERATED
{
    choice1 (1),
    choice2 (2)
}

MyTopMostSeq ::= SEQUENCE
{
    myChoice MyChoice OPTIONAL,
    myTuple MyTuple
}

MyChoice ::= CHOICE
{
    choice1 MyTuple,
    choice2 INTEGER
}

MyTuple ::= SEQUENCE
{
    fst INTEGER,
    snd INTEGER
}

END
```

```
Determinant3 DEFINITIONS ::= BEGIN

MyChoiceEnum [size 32, encoding pos-int, endianness big, encode-values]

MyTopMostSeq []
{
    myDeterminant MyChoiceEnum [],
    myChoice <myDeterminant> [],
    myTuple []
}

MyChoice <MyChoiceEnum: theDeterminant> [determinant theDeterminant]

END
```

is it expected to not be able to encode an absent `myChoice` when using the `MyChoice_NONE` as illustrated in the snippet below? Note that if we set `kind` to e.g. `choice1`, it works but it is surprising to have `choice1` or `choice2` since we specifically say that `myChoice` is absent.
```c
#include <stdio.h>
#include <assert.h>
#include "Determinant3.h"

int main(void)
{
    TMyTopMostSeq topMost = {
        .myChoice = {
            // We neither have choice1 nor choice2, so we pick NONE
            .kind = MyChoice_NONE,
            .u = {}
        },
        .myTuple = {.fst = 1, .snd = 2},
        .exist = {.myChoice = 0}
    };
    unsigned char buf[TMyTopMostSeq_REQUIRED_BYTES_FOR_ACN_ENCODING];
    BitStream bs;
    BitStream_Init(&bs, buf, sizeof buf);
    int errCode;
    flag ok = TMyTopMostSeq_ACN_Encode(&topMost, &bs, &errCode, true);
    // Prints ok = 0  errCode = 0
    printf("ok = %d  errCode = %d\n", ok, errCode);

    bs.currentByte = 0;
    bs.currentBit = 0;
    TMyTopMostSeq topMostDec;
    ok = TMyTopMostSeq_ACN_Decode(&topMostDec, &bs, &errCode);
    // Prints ok = 0  errCode = ERR_ACN_DECODE_MYTOPMOSTSEQ_MYDETERMINANT
    printf("ok = %d  errCode = %d\n", ok, errCode);
    // Decoding fails so this statement is undefined
    // printf("fst = %ld  snd = %ld\n", topMostDec.myTuple.fst, topMostDec.myTuple.snd);
}
```

maxime-esa, 2024-01-16:
I think your ACN definition is wrong for what you try to do.
As soon as you introduce the "determinant" field outside from the CHOICE, I can see why the tool has a problem with the value to put in that field when the choice field is absent. The choice determinant value must be valid, and Choice_NONE is not a valid value (encoding-wise).

The right approach would be to wrap the choice in a SEQUENCE and add the determinant there in ACN.
Something like

```
MyChoice ::= SEQUENCE {
   theChoice CHOICE { 
    ...
   }
```
and in ACN:
``` 
MyChoice [] {
    determinant [ ... ],
    theChoice [] {
       choice1 [present-when determinant == 7 ],
      ...
    }
}
```

mario-bucev, 2024-01-16:
Thank you for the clarification! I should have added that this example is inspired from `PUS-4-2.Statistics` ([ASN.1](https://github.com/maxime-esa/asn1scc/blob/21786bbdf5155c7d6aeecc409752868cb59ff26a/PUSCScalaTest/asn1-pusc-lib-asn1CompilerTestInput/service-04/PUS-4-2.asn1#L45-L60) and [ACN](https://github.com/maxime-esa/asn1scc/blob/21786bbdf5155c7d6aeecc409752868cb59ff26a/PUSCScalaTest/asn1-pusc-lib-asn1CompilerTestInput/service-04/PUS-4-2.acn#L34-L49)) hence this peculiar definition.

maxime-esa, 2024-01-16:
It is a bit different however, because in the Statistics type there is no optionality due to the subtyping (the field is marked as always present):

[image]

usr3-1415, 2024-03-03:
I've attempted to reproduce the behavior you described but was unable to do so. I ran asn1scc with the following command:
```
asn1scc -c -ACN -atc -o c_out/ -typePrefix asn1scc_ a.asn1 a.acn
```
Then, I replaced `mainprogram.c` with the code you provided and received this output:
```
ok = 0  errCode = 65
ok = 0  errCode = 7
```
Clearly, the first line (`ok=0`) indicates that the encoding has failed with error code 65. Upon checking the `a.h` file, we find the following definition:
```c
#define ERR_ACN_ENCODE_MYTOPMOSTSEQ_MYDETERMINANT_UNINITIALIZED		65  /**/
```
This error is returned during the encoding of `MyTopMostSeq` SEQUENCE because the ACN inserted field `myDeterminant` cannot be initialized, which is the expected behavior since the choice hasn't been initialized.

Could you please verify with the latest version of asn1scc and see if the problem persists on your side? If the issue no longer exists, could you please close the ticket?

maxime-esa, 2024-03-04:
This was related to #289 so it is indeed probably fixed by the same commit

closed 2024-07-16 by maxime-esa