esa/asn1scc #292: Failure when encoding/decoding messages in Streaming Mode with large Bit Strings (and smaller BitStream buffers)

ajuncosa, 2024-02-22:
Hello!

I am using asn1scc to encode and decode messages, whose total size is unaligned at bit level. I am using Streaming Mode so that the `bitstream_fetch_data_if_required()` and `bitstream_push_data_if_required()` functions will be called (as well as my implementations of `fetchData` and `pushData`).

The last member of some of my ASN structures is a `BIT STRING` with big (and unaligned to byte) sizes. This translates to a byte array member `arr` in the C structures. Then, when my BitStream's buffer size (i.e.` count` member) is large enough to store this `arr` member, everything works as expected. However, if the buffer size is smaller than the size required to store this byte array, the encode and decode functions fail (I have tried uPER and ACN encodings, but the problem remains the same in both).

I have realised that the problem seems to lie in the `BitSotream_AppendByteArray()` and `BitStream_ReadByteArray()` functions in `asn1crt_encoding.c`. During encoding, when the `BitStream_AppendByteArray()` is called, there is a line that performs the following check:

```
if ((pBitStrm->currentByte + arr_len)*8 + cb > pBitStrm->count*8)
    return FALSE;
```

Of course, when the `pBitStrm->count` is smaller than the size of the `arr_len` (or when it is slightly bigger but the `currentByte` is not 0), this check will always fail, and `pushData()` will never be called (it is called in the lines below that `return`). Therefore, my message will be partially encoded and will be missing the entire BIT STRING part. If I remove this check, everything works fine again.

However, when decoding, the same check is performed in `BitStream_ReadByteArray()`, and I am unsure how to fix it, as simply removing the check causes other errors in decoding (because `fetchData` is still not called, and therefore the decoding fails because it is unable to decode that final BIT STRING).

For example, if one of the `BIT STRING`s is 900 bits, i.e. 112.5 bytes, and my BitStream's buffer is 100 bytes. Let´s say that at the time of calling `BitStream_ReadByteArray()`, the BitStream's `currentByte` is 11, `currentBit` is 6 and `count` is 100. Then the function will return `FALSE` in that check. On the other hand, if I remove the check, it will still not call `fetchData()` because it does not meet the `fetch_data_if_required()` requirements (that the BitStream's `currentByte` must be equal to `count` in order to fetch).

Is this the intended behaviour or is it a bug? Do you have any ideas that could point us in the right direction? Thanks in advance!

usr3-1415, 2024-03-03:
To determine whether this behavior is due to an unintended bug in asn1scc or an aspect of your implementation, we would need more context.

Please, provide a **minimal** ASN.1/ACN grammar that reproduces this issue, your implementations of fetchData and pushData, and instructions on how exactly you invoke asn1scc and then gcc to compile and run your application. 

This will allow us to focus on the core issue without the complexity of debugging the entire application.

ajuncosa, 2024-03-07:
Sure, here is a simplified example of how our logic works.

Say we had a `message_definitions_example.asn1`  file with the following content:

```
Message-Definitions-Example DEFINITIONS AUTOMATIC TAGS ::= BEGIN

-- IF-EX-FILE Interface ICD

SomeMessage ::= SEQUENCE {
    someData INTEGER(0..2),
    someOtherData BIT STRING(SIZE(32))
}

END
```

And then this main.cpp file (including our `fetchData()` and `pushData()` implementations):

```
#include <iostream>
#include <bitset>
#include <cstring>
#include "message_definitions_example.h"

#ifndef BUFFER_SIZE
    # define BUFFER_SIZE 5
#endif

std::string binaryDataToAsciiString(std::string binaryData);
std::string asciiStringToBinaryData(std::string asciiString);
void encodeMessage(BitStream *bitStream, uint8_t someData, char *someOtherData);
SomeMessage decodeMessage(BitStream *bitStream);

int main(void)
{
    std::string encodedMessage; // we do not actually use a string (we use a socket), this is for simplification

    /* Step 1: Initialize some stuff */
    BitStream outputBitstream;
    BitStream inputBitstream;
    unsigned char inBuffer[BUFFER_SIZE];
    unsigned char outBuffer[BUFFER_SIZE];

    BitStream_AttachBuffer(&inputBitstream, inBuffer, BUFFER_SIZE);
	inputBitstream.pushDataPrm = nullptr;
	inputBitstream.fetchDataPrm = &encodedMessage;

    BitStream_AttachBuffer(&outputBitstream, outBuffer, BUFFER_SIZE);
	outputBitstream.pushDataPrm = &encodedMessage;
	outputBitstream.fetchDataPrm = nullptr;

    // this is added so that fetchData will be called first thing when deserializing for the 1st time
    inputBitstream.currentByte = 0;
    inputBitstream.count = 0;

    /* Step 2: Encode a message */
    uint8_t someData = 0;
    unsigned char someOtherData[4] = {255, 255, 255, 255}; // we will just fill the bit string with 1s
    encodeMessage(&outputBitstream, someData, reinterpret_cast<char*>(someOtherData));

    std::cout << "ENCODED MESSAGE: " << std::endl;
    std::cout << binaryDataToAsciiString(encodedMessage) << std::endl;

    /* Step 3: Decode the same message as was encoded before */
    encodedMessage = asciiStringToBinaryData("0011111111111111111111111111111111000000");
    SomeMessage decodedMessage = decodeMessage(&inputBitstream);
    std::cout << "DECODED MESSAGE: " << std::endl;
    std::cout << "    someData: " << (uint16_t)(decodedMessage.someData) << std::endl;
    std::cout << "    someOtherData: " << binaryDataToAsciiString(std::string(reinterpret_cast<char*>(decodedMessage.someOtherData.arr), sizeof(decodedMessage.someOtherData.arr))) << std::endl;
[... 101 lines cut ...]
    }
    return s;
}

std::string asciiStringToBinaryData(std::string asciiString)
{
    std::string s;

    for (int i = 0; i < asciiString.length(); i+=8)
    {
        s.push_back((char)(std::bitset<8>(asciiString, i, 8).to_ulong()));
    }
    return s;
}
```

This programme will create an input `BitStream` and a separate output `BitStream`, each with their associated buffers, and it will attach a string to their `fetchDataPrm` and `pushDataPrm`, respectively (instead of a string, our real code will use a class that acts as a TCP socket wrapper). Next, it will simply encode a simple message, and then decode an equivalent message, and print the output to console.

Then, we invoke asn1scc as follows: `asn1scc -c --acn-enc --slim --out asn_files message_definitions_example.asn1` (even though there is no ACN in this example, we do use it in our real code, but it does not affect this issue). Next, we can compile the small programme like: `g++ main.cpp asn_files/*.c -I asn_files/ -DASN1SCC_STREAMING -DBUFFER_SIZE=5`.

This will work as expected, it will output a string representation of the bytes that were encoded, and it will then print the contents of the decoded structure. Note that, in this example, the ASN structure requires 5 bytes to be encoded: 2 bits for the `someData` member, and 32 bits for the `someOtherData` member (the last 6 bits will just be 0 padding). Therefore, as long as `BUFFER_SIZE` >= 5, it works fine.

Now, if we compile the programme again, but using `-DBUFFER_SIZE=4` (or any smaller number), the message will be partially encoded (or decoded), apparently due to the previously identified conditions in the `BitStream_AppendByteArray()` and `BitStream_ReadByteArray()` functions from `asn1crt_encoding.h` file.

Hope this is useful and thank you for your response!

usr3-1415, 2024-03-10:
Thanks for your detailed feedback. I 've made a modification in the BitStream_ReadByteArray function that should address the issue. Please have a look and let me know.

closed 2024-07-16 by maxime-esa