esa/asn1scc #293: Function optimisation in the RTL

deme3, 2024-02-29:
Hello,

I've been using asn1scc for a while, and honestly my binary wasn't really updated for a few years until now (because I found out I had a bug caused by a missing null terminator), but now I'm having issues with my existing (non-auto generated) code. I'm using the `ByteStream_Init` function in my code before a call to the `<ASN1 Name>_XER_Encode` function (passing the ByteStream into that).

The issue basically starts with 015a7ca, which has implemented an optimisation of the generated RTL by removing functions that remain unused after code generation.

However, if I want to use functions externally that are not necessarily needed during generation, now I have to manually edit the files after generation, which is not ideal.

I believe it would be a good idea to maybe add a flag to specify which functions to always include (which basically modifies `AlwaysPresentRtlFuncNames`, if I correctly understand the code behind this).

Currently, I run the compiler with the following flags:

```bash
asn1scc -typePrefix ASN_Structure_ -icdAcn structure_acn_spec.html -XER -c structure.asn1 structure.acn
```

My proposal for this would be to add a flag `--include-func "ByteStream_Init"` which can be used multiple times in an append mode. Thoughts or alternative solutions I might employ? (other than reverting to before that commit)

Thanks

usr3-1415, 2024-03-03:
Implemented as proposed. Users can now specify the RTL functions to be generated, even if they are not used by the RTL, using the argument --include-func "ByteStream_Init" (or -if ByteStream_Init). The argument --include-func (or -if) can be repeated multiple times to include additional functions. 
Please review the code, and if everything is satisfactory, kindly close the issue.

deme3, 2024-03-03:
Looks good to me, thanks :)

closed 2024-03-03 by deme3