esa/asn1scc #312: BOOLEAN with true-value check swapped

escmmo, 2024-06-26:
Present in asn1scc 4.5.2.0
Not present in asn1scc 4.5.0.10 (did not test other versions)

Compiler flags:  --auto-test-cases --c-lang --acn-enc --word-size 8 --slim
All tests pass in both versions.

When using type BOOLEAN with ACN true-value constraint, the generated code swaps true and false value checks when compared to previous version, thereby producing the opposite result.

Example:
```asn1
BoolTest DEFINITIONS AUTOMATIC TAGS ::= BEGIN
BoolFlag ::= BOOLEAN
END
```

```acn
BoolTest DEFINITIONS ::= BEGIN
BoolFlag [true-value '1'B]
END
```

Generated code snippet in 4.5.2.0 (see lines marked with "// <---------"):
```c
flag BoolFlag_ACN_Encode(const BoolFlag* pVal, BitStream* pBitStrm, int* pErrCode, flag bCheckConstraints)
{
...
                static byte true_data[] = {0x7F};   // <---------
                static byte false_data[] = {0x80};  // <---------
                byte* tmp = (*(pVal)) ? true_data : false_data;
                BitStream_AppendBits(pBitStrm, tmp, 1);
...
}

flag BoolFlag_ACN_Decode(BoolFlag* pVal, BitStream* pBitStrm, int* pErrCode)
{
...
                static byte tmp[] = {0x7F};  // <---------
                ret = BitStream_ReadBitPattern(pBitStrm, tmp, 1, pVal);
            *pErrCode = ret ? 0 : ERR_ACN_DECODE_BOOLFLAG;
...
}
```

Code snippet in 4.5.0.10:
```c
flag BoolFlag_ACN_Encode(const BoolFlag* pVal, BitStream* pBitStrm, int* pErrCode, flag bCheckConstraints)
{
...
                static byte true_data[] = {0x80};  // <---------
                static byte false_data[] = {0x7F};  // <---------
                byte* tmp = (*(pVal)) ? true_data : false_data; 
                BitStream_AppendBits(pBitStrm, tmp, 1);
...
}

flag BoolFlag_ACN_Decode(BoolFlag* pVal, BitStream* pBitStrm, int* pErrCode)
{
...
                static byte tmp[] = {0x80};  // <---------
                ret = BitStream_ReadBitPattern(pBitStrm, tmp, 1, pVal);
            *pErrCode = ret ? 0 : ERR_ACN_DECODE_BOOLFLAG;
...
}
```

maxime-esa, 2024-06-26:
I think this was fixed in the latest commits in master (@usr3-1415 to confirm?). The version update and re-packaging will be done in a couple of days.

usr3-1415, 2024-06-28:
Yes this is confirmed!

maxime-esa, 2024-07-13:
fixed in release https://github.com/maxime-esa/asn1scc/releases/tag/4.5.2.1

closed 2024-07-16 by maxime-esa