esa/asn1scc #316: ENCODED BY Isuue

ErfanBabakhani, 2024-09-07:
hello when I try compile this code I give tc.asn:6:4: error: missing R_PAREN at 'ENCODED' error : (Doe's your compiler support ENCODED BY constraint?)
(I tried the given code by OSS NOKALVA compiler and it has no problem with it)

RectangleModule1 DEFINITIONS AUTOMATIC TAGS ::= BEGIN

Erf ::=INTEGER (0..1)
Ex1 ::= OCTET STRING (
    CONTAINING Erf 
    ENCODED BY {joint-iso-itu-t(2) asn1(1)
    packed-encoding(3) basic(0) aligned(0)})

enc-PER-Aligned OBJECT IDENTIFIER ::=
    {joint-iso-itu-t(2) asn1(1)
    packed-encoding(3) basic(0) aligned(0)}

Ex2 ::= BIT STRING (
    CONTAINING IA5String (SIZE(0..8))
    ENCODED BY enc-PER-Aligned)

END

maxime-esa, 2024-09-07:
No, asn1scc does not support this directive. The encoders are specified with the command line (either uPER, ACN or XER). aPER could probably be achieved using ACN using the `align-to` keyword...

ErfanBabakhani, 2024-09-08:
What about informtion object class ?
I think it's better to clarify types and features your compiler support and wich not supported in your doc.
https://taste.tuxfamily.org/wiki/index.php?title=Technical_topic:_ASN.1_-_An_introduction_to_ACN

thank

maxime-esa, 2024-09-08:
You can find this list of features here: https://taste.tuxfamily.org/wiki/index.php?title=Technical_topic:_ASN1SCC_-_ESA%27s_ASN.1_Compiler_for_safety-critical_embedded_platforms

closed 2024-09-23 by maxime-esa