esa/asn1scc #321: IEEE754-1985-32 test cases use double precision

Sedictious, 2025-02-18:
Hello!

When defining a minimal case with single precision floats in ACN `T-float32 [encoding IEEE754-1985-32]`, when the equivalent range is defined `T-float32 ::= REAL(-6.28318..6.28318)` the generated test cases seem to be using double precision values.

As an example, in the generated test use case, I get:

```cpp
static T_float32 tc_data; 
tc_data = -6.2831799999999997652E+000;
```

which doesn't fit into the mantissa of single precision floats and as a result the test case fails. When instead `tc_data` is set to something that fits into single precision float without rounding error (e.g. `-6.2831799`), the test case passes as expected

I tried to take a brief look at the compiler (only statically so couldn't verify any of this), and it looks like in `emitTestCaseAsFunc`, `sInitializeTcData` is only calculated based on the `REAL` type instead of considering the difference in precision from the acn encodings.

Could you confirm that this is indeed an error and not something that can be fixed purely on modifying the grammar definitions?

Thank you in advance :)

Sedictious, 2025-02-18:
Thank you for your reply @maxime-esa !

Even with the flag the unit test fails which makes sense because as you say, some precision is lost when casting `6.2831799999999997652E+000` to a single precision float.

But this still results in failing unit tests, correct? I didn't see any difference in the generated initial values in the test cases since, if my understanding is correct, everything gets mapped to `REAL` and there is no distinction when generating the test cases.

In this case, if my understanding is correct, we are casting `6.2831799999999997652` to a `float` - the actual value of which should be `6.28318023681640625` which is out of range according to the asn definitions and hence the test will fail. I guess this is intended behaviour if everything is represented as a REAL internally and as such, cases will fail when testing with double precision .

I was mainly wondering whether it would be desirable to instead of ignoring those test cases, take the width/ACN type into account when defining `tc_data`

maxime-esa, 2025-02-19:
I am not sure...

In this example the problem is that the range of the type is not compatible with the encoding constraint (IEEE-32 bits representation). 

But then ... it looks quite right to me that the test fails in that case, as it actually spots this issue, allowing the user to fix it. No?

If you keep this range but don't have a test that detects the issue, then the actual user code (not the test case) may well be assigning this value to the variable and get the error at runtime when trying to encode?

@usr3-1415 any thoughts on this?

Sedictious, 2025-02-19:
My understanding is that if the user defines a short float, they should expect to use it only to encode short floats. If they are trying to encode a double into a code that is a user failure. So I agree, I would expect the test case to fail but I'm not sure if there should be failing tests because the user would have to work within the constraints. If for some reason there is an implicit cast from a double to a float, the compiler should be complaining in any case

Because if I understand it correctly, with the current set-up the only fix would be to not use single precision at all which might not be desirable in many cases where the precision might be enough to meet the requirements and there is a restriction on the bitstream length

Sedictious, 2025-02-19:
Ok it makes sense, I understand that the value will have an inherent error based on the representation and the constraints could indeed be redefined. The only thing that I find a bit unexpected is the tests fail whereas I would expect this exact range determination to happen on the user level; if the user wants to restrict the size of the mantissa, they must know about the internal conversion error when calculating the necessary representation and it seems a bit weird to enforce this from the grammar tests and for the tests to not take the exact encoding representation into account.

But in any case, I agree that if this intended behaviour then the grammar change can be sufficient and at least on my side, the issue can be closed

Thanks again!

usr3-1415, 2025-02-23:
Perhaps `asn1scc` could generate a warning when an ASN.1 `REAL` constant literal cannot be precisely represented in the specified ACN encoding (e.g., `IEEE754-1985-32` or `IEEE754-1985-64`). For example, if a user defines a constant `6.28318` with `IEEE754-1985-32` encoding, the compiler could emit:

```
Warning: The constant 6.28318 cannot be precisely represented in IEEE754-1985-32 encoding.
The closest representable values are 6.2831797 and 6.2831802.
```

This would alert users to potential precision issues during compilation, allowing them to adjust their ASN.1 specifications accordingly.

@maxime-esa, @Sedictious: What are your thoughts on this approach?

Sedictious, 2025-02-23:
@usr3-1415 Personally, from a user's perspective, this would be very helpful indeed. My main issue was that it wasn't clear why the tests were failing without taking a look at the generated code and even then, it wasn't clear whether the behaviour was intentional or not.

If such a warning gets thrown, it is immediately clear that the problem lies in the grammar definition

usr3-1415, 2025-03-03:
Thanks for the feedback, @Sedictious! Based on this discussion, I have implemented functionality that prints a warning when an ASN.1 REAL value cannot be precisely represented in IEEE754-1985-32 encoding. This should make it clearer when precision loss occurs due to ACN constraints.

Closing this issue as resolved. Feel free to reopen or open a new issue if further refinements are needed.

closed 2025-03-03 by usr3-1415