esa/asn1scc #322: Invalid ACN C code generated for optional Sequence containing size determinant

reliable-apalla, 2025-02-19:
Hi,

I seem to have discovered an edge case in the C code generation logic where a hyphenated field name is not properly converted to its C-safe name, resulting in invalid syntax.

The issue can be reproduced using the following ASN.1 and ACN source:

```asn1
TEST DEFINITIONS AUTOMATIC TAGS ::= BEGIN

Parent ::= SEQUENCE {
        field-a                 INTEGER(0..15),
        child-entry             Child OPTIONAL
    }

    Child ::= SEQUENCE {
        payload             OCTET STRING(SIZE(0..255))
    }

END
```

```
TEST DEFINITIONS ::= BEGIN

Parent [] {
    field-a [],
    child-entry [present-when field-a >= 2]
}

Child [] {
    -- Explicitly-defined size field.
    payload-size INTEGER [encoding pos-int, size 8],

    payload [size payload-size]
}

END
```

The generated C looks like this (note the `child-entry` in the second if-statement):
```C
flag Parent_ACN_Encode(const Parent* pVal, BitStream* pBitStrm, int* pErrCode, flag bCheckConstraints)
{
   ...
                ret = pVal->exist.child_entry == ((pVal->field_a >= 2) ? 1 : 0);
	        *pErrCode = ret ? 0 : ERR_ACN_ENCODE_PARENT_CHILD_ENTRY_PRESENT_WHEN_EXP_FAILED;
	        /*Encode child_entry */
	        if (pVal->exist.child_entry) {
	        	if (pVal->exist.child-entry) {
	        	    Parent_child_entry_payload_size_is_initialized = TRUE;
	        	    Parent_child_entry_payload_size = pVal->child_entry.payload.nCount;
	        	}
	        	if (ret) {
	        	    /*Encode Parent_child_entry_payload_size */
   ...
}
```

I'm currently compiling with v4.5.2.4. Any insights would be greatly appreciated!

maxime-esa, 2025-02-19:
Thanks for reporting, we will fix this.

usr3-1415, 2025-02-19:
The bug should be fixed in the latest commit (version 4.5.2.5). Please verify that the issue is resolved, and if everything looks good, feel free to close it.

reliable-apalla, 2025-02-20:
Can confirm the bug is no longer present in v4.5.2.5, thanks!

closed 2025-02-20 by reliable-apalla