esa/asn1scc #325: code generation error when generating Xer

jmmery-dev, 2025-02-25:
Hi, 

When generating XER encoder, some  xxx_XER_Decode_aux and xxx_XER_Encode_aux have a problem with variable definition ( declaring int i2, using i1 )

**for decode :**
flag TableItineraireAg_XER_Decode_aux(TableItineraireAg* pVal, const char* xmlTag, ByteStream* pByteStrm, int* pErrCode)
{
    flag ret = TRUE;
	int i2;
	(void)pVal;
	(void)xmlTag;
	(void)pByteStrm;
	(void)pErrCode;

	/* SEQUENCE OF Decode*/
	ret = Xer_DecodeComplexElementStart(pByteStrm, xmlTag, NULL, pErrCode);
	if (ret) {
	    i1 = 0;
	    while(ret && !Xer_NextEndElementIs(pByteStrm, xmlTag))
	    {
		    ret = ParametresItineraireAg_XER_Decode_aux((&(pVal->arr[i1])), "ParametresItineraireAg", pByteStrm, pErrCode);
		    i1++;
	    }
	    if (ret) {
	        ret = Xer_DecodeComplexElementEnd(pByteStrm, xmlTag, pErrCode);
	    }
	}
	*pErrCode = (ret && i1 == 16) ? 0 : ERR_XER_DECODE_TABLEITINERAIREAG;

	return ret;
}
**for encode **
`flag ListeParametresAg_XER_Encode_aux(const ListeParametresAg* pVal, const char* xmlTag, ByteStream* pByteStrm, int* pErrCode, flag bCheckConstraints)
{
    flag ret = TRUE;
	(void)pVal;
	(void)xmlTag;
	(void)pByteStrm;

	int i2;
	ret = bCheckConstraints ? ListeParametresAg_IsConstraintValid(pVal, pErrCode) : TRUE ;
	if (ret) {
	    /* SEQUENCE OF Encode*/
	    ret = Xer_EncodeComplexElementStart(pByteStrm, xmlTag, NULL, pErrCode, 0);
	    *pErrCode = ret ? 0 : ERR_XER_ENCODE_LISTEPARAMETRESAG;
	    if (ret) {
	        for(i1=0;((i1 < pVal->nCount) && ret);i1++)
	        {
	    	    ret = ConfigurationBloc_XER_Encode_aux((&(pVal->arr[i1])), "ConfigurationBloc", pByteStrm, pErrCode, FALSE);
	        }
	        if (ret) {
	            ret = Xer_EncodeComplexElementEnd(pByteStrm, xmlTag, pErrCode, 0);
	            *pErrCode = ret ? 0 : ERR_XER_ENCODE_LISTEPARAMETRESAG;
	        }
	    }
    }

    return ret;
}

my options for generating :  -XER  -if ByteStream_Init -c , and my version of asn1scc 4.5.2.4

After replacing i2 by i1 in declaration everything seems to work.

jmmery-dev, 2025-03-03:
the asnfile :

jmmery-dev, 2025-03-03:
the asn definition (defined in BUG.asn):

```
SYSTEM-INTERFACES DEFINITIONS AUTOMATIC TAGS ::= BEGIN

SYSTEM-TOR ::= SEQUENCE {
    value	BOOLEAN,
    name	IA5String(SIZE(10))
}

SYSTEM-INPUTS ::= SEQUENCE SIZE( 4096 ) OF SYSTEM-TOR

SYSTEM-OUTPUTS ::= SEQUENCE SIZE( 1536 ) OF SYSTEM-TOR

END

```
the compile line :
`asn1scc -XER -c -if ByteStream_Init BUG.asn`

the code generated ( piece of ):

```
flag SYSTEM_INPUTS_XER_Encode_aux(const SYSTEM_INPUTS* pVal, const char* xmlTag, ByteStream* pByteStrm, int* pErrCode, flag bCheckConstraints)
{
    flag ret = TRUE;
	(void)pVal;
	(void)xmlTag;
	(void)pByteStrm;

	int i2;
	ret = bCheckConstraints ? SYSTEM_INPUTS_IsConstraintValid(pVal, pErrCode) : TRUE ;
	if (ret) {
	    /* SEQUENCE OF Encode*/
	    ret = Xer_EncodeComplexElementStart(pByteStrm, xmlTag, NULL, pErrCode, 0);
	    *pErrCode = ret ? 0 : ERR_XER_ENCODE_SYSTEM_INPUTS;
	    if (ret) {
	        for(i1=0;((i1 < 4096) && ret);i1++)
	        {
	    	    ret = SYSTEM_TOR_XER_Encode_aux((&(pVal->arr[i1])), "SYSTEM-TOR", pByteStrm, pErrCode, FALSE);
	        }
	        if (ret) {
	            ret = Xer_EncodeComplexElementEnd(pByteStrm, xmlTag, pErrCode, 0);
	            *pErrCode = ret ? 0 : ERR_XER_ENCODE_SYSTEM_INPUTS;
	        }
	    }
    }

    return ret;
}

```
the version:
`asn1scc version 4.5.2.6`

usr3-1415, 2025-03-08:
Fixed in the latest commit.
Thanks for reporting it.

closed 2025-03-08 by usr3-1415