esa/asn1scc #330: Possible invalid memory access in BitStream_ReadByte

Sedictious, 2025-04-03:
Hello!

I wanted to report an issue with a possible invalid memory access in `BitStream_ReadByte`. Namely, the following code gets generated:

```c
flag BitStream_ReadByte(BitStream* pBitStrm, byte* v) 
{
    int cb = pBitStrm->currentBit;
    int ncb = 8 - pBitStrm->currentBit;
    *v= (byte)(pBitStrm->buf[pBitStrm->currentByte++]  << cb);

    if (cb) {
        *v |= (byte)(pBitStrm->buf[pBitStrm->currentByte] >> ncb);
    }

    return pBitStrm->currentByte*8+pBitStrm->currentBit<=pBitStrm->count*8;
}
```

In this snippet, `pBitStrm->buf[pBitStrm->currentByte]` may point to an invalid address if it exceeds the limits indicated by `pBitStrm->count`. In such case, the condition is checked in the return condition with `pBitStrm->currentByte*8+pBitStrm->currentBit<=pBitStrm->count*8` so the return code would in that case would return `false` as expected.

However, since `buf[pBitStrm->currentByte]` may attempt to access invalid memory, wouldn't the bound check need to be carried out before attempting to read instead? The condition `(pBitStrm->currentByte + (pBitStrm->currentBit > 0)) * 8 + pBitStrm->currentBit <= pBitStrm->count * 8` could be checked at the beginning of the function and exit immediately if it is not met to avoid the possible invalid memory access.

usr3-1415, 2025-04-06:
Thanks a lot for reporting this!

You were right — the original implementation could access memory beyond the bounds of the buffer before the condition was checked. We've now added a proper bounds check before accessing any bytes from the buffer. This should eliminate the possibility of invalid memory access.

If everything looks good on your side, feel free to go ahead and close the issue — otherwise let me know if you spot anything else!

Sedictious, 2025-04-07:
Thanks a lot for the fix @usr3-1415  ! Looks good on my side so closing the issue

closed 2025-04-07 by Sedictious