esa/asn1scc #347: Cant use custom type in present-when

dsabala, 2025-09-09:
Hi

This works:

rep2.asn
```
Mytest DEFINITIONS AUTOMATIC TAGS ::= BEGIN

UInt16 ::= INTEGER (0..65535)
UInt32 ::= INTEGER (0..4294967295)

StatusValues ::= ENUMERATED {
  success(0),
  fail(1)
}

IdentityValues ::= ENUMERATED {
  bootloader(0),
  application-bank-1(1),
  application-bank-2(2)
}

Timestamp32 ::= UInt32 -- seconds since Unix epoch

IdentityPayload ::= SEQUENCE {
  identity               IdentityValues,
  version-major          UInt32,
  version-minor          UInt32,
  version-patch          UInt32,
  build-unix-timestamp   Timestamp32
}

IdentityResponse ::= SEQUENCE {
  status UInt32,
  payload IdentityPayload OPTIONAL
}

END
```

rep2.acn
```
Mytest DEFINITIONS ::= BEGIN

UInt16         [encoding pos-int, size 16, endianness little];
UInt32         [encoding pos-int, size 32, endianness little];

StatusValues   [encoding pos-int, size 16, endianness little];
IdentityValues [encoding pos-int, size 16, endianness little];

IdentityResponse [] {
  status [],
  payload [present-when (status == 0)]
};

END
```

```bash
./asn1-docker.sh --acn-enc -c -o icdpoc --icdacn docs/xd.html -uPER rep2.asn rep2.acn
```

Judging by HTML documentation it works, but it creates new type called `Mytest-IdentityResponse-payload` instead of using already exising type with cleaner name `IdentityPayload`.

Furthermore I cant change type of status field to enumerated to make auto-generated code more safe and easy to use

```
Mytest DEFINITIONS AUTOMATIC TAGS ::= BEGIN

UInt16 ::= INTEGER (0..65535)
UInt32 ::= INTEGER (0..4294967295)

StatusValues ::= ENUMERATED {
  success(0),
  fail(1)
}

IdentityValues ::= ENUMERATED {
  bootloader(0),
  application-bank-1(1),
  application-bank-2(2)
}

Timestamp32 ::= UInt32 -- seconds since Unix epoch

IdentityPayload ::= SEQUENCE {
  identity               IdentityValues,
  version-major          UInt32,
  version-minor          UInt32,
  version-patch          UInt32,
  build-unix-timestamp   Timestamp32
}

IdentityResponse ::= SEQUENCE {
  status StatusValues,
  payload IdentityPayload OPTIONAL
}

END
```

```
Mytest DEFINITIONS ::= BEGIN

UInt16         [encoding pos-int, size 16, endianness little];
UInt32         [encoding pos-int, size 32, endianness little];

StatusValues   [encoding pos-int, size 16, endianness little];
IdentityValues [encoding pos-int, size 16, endianness little];

IdentityResponse [] {
  status [],
  payload [present-when (status == success)]
};

END
```

Because it outputs

```
./asn1-docker.sh --acn-enc -c -o icdpoc --icdacn docs/xd.html -uPER rep2.asn rep2.acn
rep2.acn:11:25: error: Invalid reference 'status'
```

maxime-esa, 2025-09-09:
Yes I was about to suggest the `CHOICE` alternative indeed.

I've flagged the need to support `present-when <enumeratedtype==value>` as a feature request but with low priority for the moment.

Regarding the issue you have with the ICD, could you try opening the icd_`new`.html file instead of just `icd.html` ?

dsabala, 2025-09-09:
Yes, indeed `icd_new.html` no longer contains these redundant data types. 

It looks like `asn1scc` might be a tool that fits my current needs, I will give this library a Github star, thank you very much :)